LVD-AC/Marlin-AC #18: G33 with eccentric probe, not probing with probe offsets

AlbertPhan, 2017-08-21:
- Description: When I used the G33 P2 T V0 E  Auto delta calibration command, it probes points without adding offsets to the nozzle to where the probe is. I use a semi-auto probe which i need to deploy/attach myself on my delta (kossel XL). I have AUTO_BED_LEVELING_BILINEAR defined. I cannot use UBL as I get compile errors relating to memory size (probably too big for my printrboard).
- Expected behaviour: Probing with respect to the probe.
- Actual behaviour: Probes with respect to the nozzle.
- Steps to reproduce:
Use G33 P2 T V0 E with or without E or P1 instead of P2 produces the same results.

Is this intended behaviour? I can help with any testing. 

I am in the process of creating a probe holder which goes directly under the nozzle.

[Configuration.zip](https://github.com/LVD-AC/Marlin-AC/files/1240130/Configuration.zip)

Using M111 S32: 

SENDING:**G33 P1 V0**
G33 Auto Calibrate
reset_bed_level
  current_position=(0.00, 0.00, 275.00) : setup_for_endstop_or_probe_move
  current_position=(0.00, 0.00, 275.00) : set_probe_deployed
deploy: 1
do_probe_raise(20.00)
>>> do_blocking_move_to(0.00, 0.00, 275.00)
  destination=(0.00, 0.00, 275.00) : set_destination_to_current
  destination=(0.00, 0.00, 275.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 275.00) : danger zone move
  current_position=(0.00, 0.00, 275.00) : >>> home_delta
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, 390.00, 0.00)
  current_position=(0.00, 285.00, 285.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, -5.00, 0.00)
  current_position=(0.00, 285.00, 285.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, 10.00, 10.00)
  current_position=(0.00, 285.00, 285.00) : sync_plan_position
<<< do_homing_move(X)
endstop_adj:
>>> do_homing_move(X, -0.10, 0.00)
  current_position=(0.00, 285.00, 285.00) : sync_plan_position
<<< do_homing_move(X)
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, 390.00, 0.00)
  current_position=(-0.10, 0.00, 285.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, -5.00, 0.00)
  current_position=(-0.10, 0.00, 285.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, 10.00, 10.00)
  current_position=(-0.10, 0.00, 285.00) : sync_plan_position
<<< do_homing_move(Y)
endstop_adj:
>>> do_homing_move(Y, -0.10, 0.00)
  current_position=(-0.10, 0.00, 285.00) : sync_plan_position
<<< do_homing_move(Y)
<<< homeaxis(Y)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, 412.50, 0.00)
  current_position=(-0.10, -0.10, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, -5.00, 0.00)
  current_position=(-0.10, -0.10, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, 10.00, 10.00)
  current_position=(-0.10, -0.10, 0.00) : sync_plan_position
<<< do_homing_move(Z)
endstop_adj:
>>> do_homing_move(Z, -0.10, 0.00)
  current_position=(-0.10, -0.10, 0.00) : sync_plan_position
<<< do_homing_move(Z)
<<< homeaxis(Z)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -130.00
 soft_endstop_max = 130.00
> home_offset[X] = 0.00
  current_position=(0.00, -0.10, -0.10) :
<<< set_axis_is_at_home(X)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = -130.00
 soft_endstop_max = 130.00
> home_offset[Y] = 0.00
  current_position=(0.00, 0.00, -0.10) :
<<< set_axis_is_at_home(Y)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 275.00
> home_offset[Z] = 0.00
  current_position=(0.00, 0.00, 275.00) :
<<< set_axis_is_at_home(Z)
  current_position=(0.00, 0.00, 275.00) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 275.00) : <<< home_delta
Checking... AC (DRY-RUN)
.Height:275.00
**>>> probe_pt(0.00, 33.00, no stow)**  it sees the probe offset here
  current_position=(0.00, 0.00, 275.00) :
>>> do_blocking_move_to(0.00, 0.00, 193.54)
  destination=(0.00, 0.00, 275.00) : set_destination_to_current
  destination=(0.00, 0.00, 193.54) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 193.54) : zone border move
  current_position=(0.00, 0.00, 193.54) : xy move
<<< do_blocking_move_to
do_probe_raise(10.00)
>>> do_blocking_move_to(0.00, 0.00, 193.54)
  destination=(0.00, 0.00, 193.54) : set_destination_to_current
  current_position=(0.00, 0.00, 193.54) : xy move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 193.54) : set_probe_deployed
deploy: 1
  current_position=(0.00, 0.00, 193.54) : >>> run_z_probe
  current_position=(0.00, 0.00, 193.54) : >>> do_probe_move
>>> do_blocking_move_to(0.00, 0.00, -285.00)
  destination=(0.00, 0.00, 193.54) : set_destination_to_current
  current_position=(0.00, 0.00, 193.54) : xy move
  destination=(0.00, 0.00, -285.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -285.00) : z lower move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 8.52) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 8.52) : <<< do_probe_move
1st Probe Z:8.52
>>> do_blocking_move_to(0.00, 0.00, 13.52)
  destination=(0.00, 0.00, 8.52) : set_destination_to_current
  destination=(0.00, 0.00, 13.52) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 13.52) : z raise move
  current_position=(0.00, 0.00, 13.52) : xy move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 13.52) : >>> do_probe_move
>>> do_blocking_move_to(0.00, 0.00, -285.00)
  destination=(0.00, 0.00, 13.52) : set_destination_to_current
  current_position=(0.00, 0.00, 13.52) : xy move
  destination=(0.00, 0.00, -285.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -285.00) : z lower move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 8.67) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 8.67) : <<< do_probe_move
  current_position=(0.00, 0.00, 8.67) : <<< run_z_probe
2nd Probe Z:8.67 Discrepancy:-0.15
do_probe_raise(10.00)
>>> do_blocking_move_to(0.00, 0.00, 30.10)
  destination=(0.00, 0.00, 8.67) : set_destination_to_current
  destination=(0.00, 0.00, 30.10) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 30.10) : z raise move
  current_position=(0.00, 0.00, 30.10) : xy move
<<< do_blocking_move_to
<<< probe_pt
>>> probe_pt(0.00, 33.00, no stow)
  current_position=(0.00, 0.00, 30.10) :
do_probe_raise(10.00)
>>> do_blocking_move_to(0.00, 0.00, 30.10)
  destination=(0.00, 0.00, 30.10) : set_destination_to_current
  current_position=(0.00, 0.00, 30.10) : xy move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 30.10) : set_probe_deployed
deploy: 1
  current_position=(0.00, 0.00, 30.10) : >>> run_z_probe
  current_position=(0.00, 0.00, 30.10) : >>> do_probe_move
>>> do_blocking_move_to(0.00, 0.00, -285.00)
  destination=(0.00, 0.00, 30.10) : set_destination_to_current
  current_position=(0.00, 0.00, 30.10) : xy move
  destination=(0.00, 0.00, -285.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -285.00) : z lower move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 8.52) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 8.52) : <<< do_probe_move
1st Probe Z:8.52
>>> do_blocking_move_to(0.00, 0.00, 13.52)
  destination=(0.00, 0.00, 8.52) : set_destination_to_current
  destination=(0.00, 0.00, 13.52) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 13.52) : z raise move
  current_position=(0.00, 0.00, 13.52) : xy move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 13.52) : >>> do_probe_move
>>> do_blocking_move_to(0.00, 0.00, -285.00)
  destination=(0.00, 0.00, 13.52) : set_destination_to_current
  current_position=(0.00, 0.00, 13.52) : xy move
  destination=(0.00, 0.00, -285.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -285.00) : z lower move
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 8.67) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 8.67) : <<< do_probe_move
  current_position=(0.00, 0.00, 8.67) : <<< run_z_probe
2nd Probe Z:8.67 Discrepancy:-0.15
do_probe_raise(10.00)
>>> do_blocking_move_to(0.00, 0.00, 18.67)
  destination=(0.00, 0.00, 8.67) : set_destination_to_current
  destination=(0.00, 0.00, 18.67) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 18.67) : z raise move
  current_position=(0.00, 0.00, 18.67) : xy move
<<< do_blocking_move_to
<<< probe_pt
.      c:+0.00
End DRY-RUN                                       std dev:0.000

ThomasM1960, 2017-08-29:
I have the same behavior
The probe offset is not used at all.
Any solution? Currently I can't use the G33 command at all.

AlbertPhan, 2017-08-29:
I haven't found a solution other than modelling a probe Mount that puts it
directly under the nozzle. Then it works fine. If you have a e3d nozzle
I'll post the model for the probe when I get back home.

ThomasM1960, 2017-08-30:
@AlbertPhan did you have an display connect to the printer. I have looked into the code and think I found the reason of the problem. I will test that ASAP. I will keep you updated.

AlbertPhan, 2017-08-31:
I do not have a display.

LVD-AC, 2017-09-01:
For delta calibration G33 (opposite to bed levelling G29) the carriages, rods, effector and thus nozzle need to be at the specified position, not the probe. If with an eccentric probe the probe runs off the bed with G33 one needs to reduce DELTA_CALIBRATION_RADIUS in Configuration.h

G33 auto calibration assumes the print bed to be flat and perpendicular to the towers (check hardware) and calibrates the printer so the movement of the effector is in a flat plane as well. G29 auto leveling assumes the effector-nozzle movement to be in a flat plane (which is the case for Cartesians but not for Deltas) and compensates for unevenness and tilt of the print bed.

closed 2017-09-05 by LVD-AC

AlbertPhan, 2017-09-05:
>If with an eccentric probe the probe runs off the bed with G33 one needs to reduce DELTA_CALIBRATION_RADIUS in Configuration.h

Thanks for the heads up. 
Here's my probe holder with no xy offsets for e3d v6 hotends for those of you who want to auto-calibrate with a manually deployed probe:
https://www.thingiverse.com/thing:2517914

Once this is installed, I got G33 working fantastically. Saved a lot of time vs manually calibrating my offsets and delta radius.