LVD-AC/Marlin-AC #26: '#define MULTIPLE_PROBING 2' leads to erroneous G33 behaviour

joris57, 2017-12-29:
When enabling 'multiple probing', G33 demonstrates erratic probing behaviour, eventually leading to a 'Probing failed' error (experienced in both 1.1.7 and 1.1.8 releases).

A characteristic of this erratic behaviour is that the nozzle does not slow down when reaching the bed when executing a G33 P1.

When  '#define MULTIPLE_PROBING 2' is disabled, behaviour is back to normal.

LVD-AC, 2017-12-29:
AFAIR 2 consecutive probes are set up to do 1 fast probe that is not taken into account followed by a retraction and a slow probe; IMO a bad option since during the fast probe steps could be lost that can not be recovered until the next home, so you gradually build up errors in consecutive probes. It was set up to resemble the homing procedure: move up quickly to find the end-stop, retract and re-trigger at slower speed for more accuracy. With homing it does not matter if steps are lost during the fast move since that tower is homed (steps set zero) just afterwards with the slow trigger; with bed probes it is quit a different story. I never use multiple probing for calibration, it is to inaccurate.

joris57, 2017-12-30:
For your consideration :

1) If G33 is incompatible with MULTIPLE_PROBING, it would be nice to have an error message to this extent to save the unwary user the time of troubleshooting this issue, trying to understand why G33 behaves erratically.
2) All other leveling mechanisms (UBL, ABL, ...) seem to cope well with MULTIPLE_PROBING. Reproducability of G30 is also ok. G33 is the exception here.
3) The issue of losing steps can anyway be mitigated by slowing down the probing speed. 
4) When issuing G33 P3 the first probe on the outer probing circle only dips once, the others twice, as expected, so there seems to be something wrong with G33 and MULTIPLE_PROBING, other than the risk of losing steps.

LVD-AC, 2017-12-30:
1) not incompatible but IMO there is no advantage only a possible cause of trouble.
4) Probing is done by lifting the nozzle up from the previous z-position. Possible cause is the delta radius being way off (big difference between z at center and z at outer circle); this should return to normal in the next iterations when the delta radius is set to a more realistic value.
2) G29 probes a square grid and moves the nozzle along; thus the z-distance between 2 consequtive probes is always small -> no problems.

joris57, 2017-12-31:
1) When the same version of Marlin on the same printer with the same configuration works fine without '#define MULTIPLE_PROBING 2' and goes completely haywire with it, I call this either incompatible or buggy.
4) Delta radius is pretty accurately set, from previous G33 results and hardcoded in the conf.
2) I'm talking, among others, about probing a 3x3 grid on a 300mmx300mm bed, so non sequitur.
3) Slow is homing speed/6, fast is homing speed. There's quite a bit of room between those.

Were you not the person complaining in a recent post that the Marlin developers were in denial?

Anyway, I've done my best in reporting the issue. If you choose to ignore it, I'll close he issue.

LVD-AC, 2017-12-31:
1: > When issuing G33 P3 the first probe on the outer probing circle only dips once, the others twice
is this what you call "completely haywire"? If not, please be a bit more precise...

3: There are no 3x3 grids in G33; 300 x 300 mm bed: square bed is a rather strange config on a delta.

> 'Probing failed' error (experienced in both 1.1.7 and 1.1.8 releases)
I'm sorry but there is no 1.1.8 release in this rep. Are we talking about the same thing?

joris57, 2017-12-31:
Look at it purely from a user's perspective : G33 probing fails whilst probing outside the probable area (including scraping the bed). The user looks several hours for the origin of this problem, to find out that when he removes the for-no-particular-reason-to-be-suspected '#define MULTIPLE_PROBING 2' option, all is well again. Never mind why the user switched on '#define MULTIPLE_PROBING 2' as nobody advised him against it.

Would you consider this a good/normal user experience?

Some last replies to to your nitpicking:

1) Read the original post : '... leading to a 'Probing failed' error', What I observe are probing attempts outside the probable area, including scraping the nozzle over the bed (which I assumed was implied by 'Probing failed').

2) You mentioned ' G29 probes a square grid and moves the nozzle along; thus the z-distance between 2 consequtive probes is always small -> no problems.' --> G29 probes are not 'always small'. The example I gave was for my CoreXY, which shows no issues with double probing over larger distances (i.e. 80 mm or more).

3) 'Most of them are disabled on delta's only UBL and Bi-linear can be used.' --> I mentioned that UBL and ABL don't seem to suffer from the issue. UBL and bi-lineair ABL are exactly those mechanisms I tried with double probing. 

4) Xmas-AC seems to be based on 1.1.8. Apologies for naming it as such. 

Since you obviously are going to argue until my patience runs out, I'm happy to comply.

LVD-AC, 2017-12-31:
> What I observe are probing attempts outside the probable area,

You need to set DELTA_CALIBRATION_RADIUS for that; the "DELTA_PROBEABLE_RADIUS" is a G29 related parameter that only applies to G29.

> including scraping the nozzle over the bed (which I assumed was implied by 'Probing failed')

not really, probing can fail for many reasons e.g. by floating over the bed and not lowering enough. Anyhow the G33 routine does not continue after the 1st probe error. So...

> When issuing G33 P3 the first probe on the outer probing circle only dips once, the others twice

this can't be a "probing fail error", the routine would stop and not continue.

> G29 probes are not 'always small'. The example I gave was for my CoreXY

I on the other hand was talking about **delta printers** that could have a DELTA_HEIGHT or a DELTA_RADIUS set wrong, which both can lead to probing errors; it is obvious that such errors can only be caused on printers that have such delta specific parameters. Having the delta parameters wrong is the most common cause of probing errors during the G33 routine. A probe error during G33 also gives the following G33 specific error message:

`"Correct delta settings with M665 and M666"
`
Did you try that? Another fix can be in increasing the Z_CLEARANCE_BETWEEN_PROBES

> I mentioned that UBL and ABL don't seem to suffer from the issue

I tried to explain that on **a delta** G29 moves are always smaller than moving from the centre to the outer diameter (G33), thus are less prone to such probing errors caused by offset delta parameters.

> Xmas-AC seems to be based on 1.1.8. 

Big difference in the G33 code, since Marlin team decided to hold off any further developments to G33. So 1.1.7, 1.1.8 are seriously lacking behind 1.1.7-AC and Xmas-AC. I have to make sure I look into the right version to look for bugs.

Naming the one when talking about the other (FW version, type of printer,... ) does not make your posts any clearer to understand, but I do my best.

prahjister, 2018-01-14:
Does this mean that this version is better for Delta printers with offset probes while using g33?

closed 2018-02-11 by LVD-AC