LVD-AC/Marlin-AC #28: G33 travelling off print bed

Corbid, 2018-02-25:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---G33 travelling off print bed
- Expected behaviour: --- When entering G33 p2 command using the G-force8-AC firmware, center and points in front of towers are probed for z height position
- Actual behaviour: --- When entering G33 p2 command, center position is probed correctly, but despite adjusting the DELTA_CALIBRATION_RADIUS, the following point doesn't adjust within the print bed area. My total print bed is 220mm in diameter, and I have moved the calibration radius to as small as 55mm, but it still runs off the bed. 

[Configuration.zip](https://github.com/LVD-AC/Marlin-AC/files/1755820/Configuration.zip)

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

LVD-AC, 2018-02-25:
`- When troubleshooting, use M502 followed by M500 to reset EEPROM to defaults.`
did you use M502 M500 after changing  DELTA_CALIBRATION_RADIUS?

Corbid, 2018-02-25:
Yep. I've continued to try playing around with the DELTA_CALIBRATION_RADIUS and DELTA_PRINTABLE_RADIUS. Basic process is to update and flash the firmware, then M502, G33 P1, M500, and then G33 P2. Earlier the first touch after center would send it into the tower. I adjusted the DELTA_PRINTABLE_RADIUS to 90.0 and the DELTA_CALIBRATION_RADIUS to 80.0. This made a successful touch in the center and first tower, then went crashing towards the second tower.

LVD-AC, 2018-02-25:
If it moves ok to the center and to one tower, something must be wrong with your setup

baconinabun, 2018-02-25:
@Corbid i found that i had similar issue but it was due to my stepper drivers being set to high, also the same sort of issue was happening because i didn't have enough airflow on to the drivers they were getting far too hot, might not be your issue but worth having a look at

GAMIX66, 2018-02-25:
bonjour jai une kossel mini avec une carte mks mini le marlin cest bien installer mais la calibration automatique fonctionne pas les bras bloque et avec repetier beaucoup de comande fonctionne pas et sa imprime dans le vide jai un interupteur commme sonde z ya til un moyen de auto calibrer avec sa desoler sa fais beaucoup de probleme est ce que vous penser que ce firmware fonctionne til avec ma config normalement svp merci davance

baconinabun, 2018-02-25:
@GAMIX66 i don't speak french and google translator is well you can imagine but could be worth a look here if your understanding of english is not bad https://github.com/FLSun3dp/FLSun-Kossel-Mini/wiki/02.-Calibrate-your-printer

baconinabun, 2018-02-25:
@LVD-AC I hope you don't mind me piping in?

baconinabun, 2018-02-25:
@Corbid this is what i have my FLSUN delta set too

[image]

Corbid, 2018-02-26:
Updated settings to what worked for baconinabun. Hard to explain movement exactly, so here's  a really short clip showing what the printer is doing. The angle taken vertically from probe point 2 to 3 seems odd....
[20180226_092516_1.zip](https://github.com/LVD-AC/Marlin-AC/files/1756894/20180226_092516_1.zip)

baconinabun, 2018-02-26:
@Corbid that missing steps sound is that psu not being aboleto handle what is going on, i had similar noise so i fitted a 360w off another printer i have and that solved that, 

I have to go out to work but when i get home i will fit my GEN-L back to Delta and see what the main is issue (y)

LVD-AC, 2018-02-26:
Eyeballing it your printer does not go to the center but is quit off (could be the video)

Corbid, 2018-02-26:
LVD-AC: Yeah, after switching from the original firmware that came from FLSUN it seemed to center a little closer to the z tower. I broke it all down today and made sure everything was as square as I could get it. Just got the belts on, but don't want to wake the family by firing it up. I'll make another run through tomorrow to see if anything has changed significantly.

baconinabun: I don't remember hearing the same sound using the old firmware, but maybe the speed was intentionally set lower. I have a couple of atx power supplies kicking around. Might try to hook one of them up if I can't get this going right.

Corbid, 2018-02-27:
Waiting on a new set of diagonal rods. Will see if that makes a difference.

baconinabun, 2018-02-27:
@Corbid i've not managed to get around to looking at anything weather has been really shite this neck of the woods, hope to try and get caught up at weekend with things

closed 2018-03-13 by LVD-AC