LVD-AC/Marlin-AC #29: Setting up for dual extruder results in a compile error (rel. G-force8-AC)

joris57, 2018-03-07:
Setting up for dual extruder results in a compile error (rel. G-force8-AC):

Config changed:
> ...
> #define MOTHERBOARD BOARD_RAMPS_13_EEB
> ...
> #define EXTRUDERS 2
> ...
> #define TEMP_SENSOR_1 5
> ...

Compile error:

> Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> sketch\Marlin_main.cpp: In function 'void gcode_G33()':
> 
> Marlin_main.cpp:5384: error: 'old_tool_index' was not declared in this scope
> 
>        #define AC_CLEANUP() ac_cleanup(old_tool_index)
> 
>                                        ^
> 
> sketch\Marlin_main.cpp:5828:16: note: in expansion of macro 'AC_CLEANUP'
> 
>          return AC_CLEANUP();
> 
>                 ^
> 
> sketch\Marlin_main.cpp:5384:53: warning: return-statement with a value, in function returning 'void' [-fpermissive]
> 
>        #define AC_CLEANUP() ac_cleanup(old_tool_index)
> 
>                                                      ^
> 
> sketch\Marlin_main.cpp:5828:16: note: in expansion of macro 'AC_CLEANUP'
> 
>          return AC_CLEANUP();
> 
>                 ^
> 
> Marlin_main.cpp:5384: error: 'old_tool_index' was not declared in this scope
> 
>        #define AC_CLEANUP() ac_cleanup(old_tool_index)
> 
>                                        ^
> 
> sketch\Marlin_main.cpp:6012:5: note: in expansion of macro 'AC_CLEANUP'
> 
>      AC_CLEANUP();
> 
>      ^
> 
> exit status 1
> 'old_tool_index' was not declared in this scope

LVD-AC, 2018-03-07:
Try the 1.1.x-AC branch; the bug should be fixed now

joris57, 2018-03-12:
1.1.x-AC branch compiles.

closed 2018-03-12 by joris57