LVD-AC/Marlin-AC #36: G33 calbration issue with FLSun delta

mahe-fi, 2019-04-22:
### Description

G33 calibration broken on FLSun delta. 

I cannot do G33 beyond P1 and P2 anymore. With P3, P5 or P10 the printer probes the center points fine, but after one or two outer points, the effector stops for a second or two, then jumps 1-2mm upwards and moves toward center digging into the bed HARD.

I was able to do the whole calibration routine P1, P2, P5, P10. Recently I had to fiddle with the bed mounts on the printer and therefore needed to re-calibrate. First time the problem appeared with P10, P3 and P5 followed on next attempts.

Printer is printing fine with only P2 calibration done, so it seems that this is not any 
systematic hardware problem.

Running G-force9.

I have changed Z-axis endstop wire to Y-axis endstop socket on the board (MKS Gen-L v1.0) because the printer's board was defective on arrival (Z-endstop socket is dead). Therefore I've had to exchange values for variables Y_MIN_PIN and Z_MIN_PIN in pins_RAMPS.h and Y_MIN_ENDSTOP_INVERTING to true in configuration.h copied from FLSun example.

Otherwise everything is stock, besides the obvious changes to DELTA_HEIGHT, DIAGONAL_ROD, DELTA RADIUS and endstop adjustments.

I suspected that the tower trim adjustments could have been the culprit since they kick in with G33 P3, and zeroed them. Didn't help.

My configuration files are attached.

### Steps to Reproduce

1. G28
2. G33 PX
3. M500

X = 1, 2, 5, 10

**Expected behavior:** 

All calibration steps should succeed

**Actual behavior:** 

See Description 

[Configuration.zip](https://github.com/LVD-AC/Marlin-AC/files/3103317/Configuration.zip)

LVD-AC, 2019-04-22:
Since it worked before it seems like a hardware problem

mahe-fi, 2019-04-22:
Yeah, I too thought that is most likely explanation. I have new board coming from china, I'll see how it works when I get it. 

I just find it strange that I can print just fine, same kinds of movements don't cause any trouble then. I mean, to my limited understanding the calibration just throws the effector around and lower's it slowly until the z-probe (or endstop switch in my case) engages? Or is there something more complicated going on, movementvise?

LVD-AC, 2019-04-22:
Probing detects if the z-endstop is engaged; while printing does not...

mahe-fi, 2019-05-01:
I have now exchanged defective mainboard to my printer. The new board came with new stepper drivers so they are have changed too.

The problem remains the same: G33 P3 or G33 P5 calibrations don't work. The efector stops after one or 
two outer probe points (the points near the center are probed OK) and then slams into the bed.

My current Configuration.h is attached

[Configuration (2).zip](https://github.com/LVD-AC/Marlin-AC/files/3134779/Configuration.2.zip)

closed 2021-04-16 by LVD-AC